illegalprime/nixos-on-arm #1: volume_key compilation failure.

deadloko, 2019-04-11:
First of all, great effort for replacing yocto with nix build system. I have a custom board based on toradex apalis imx6 SOC, so I wanted to try if I can run nix on it. I have written configuration files based on rpi examples, but when I tried to build sd image, i ran into compilation error. 

Error log:
```bash
nix build -f . \
  -I nixpkgs=nixpkgs \
  -I machine=machines/toradex_apalis_imx6 \
  -I image=images/minimal

builder for '/nix/store/mwm4n5i1i0qkrdaqpd578l7z4basgqgy-volume_key-0.3.11-armv7l-unknown-linux-gnueabihf.drv' failed with exit code 2; last 10 log lines:
  libtool: compile:  armv7l-unknown-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I/nix/store/dgijzk00lch1wg43c7ymj9qnbg2jv6gn-util-linux-2.33.1-armv7l-unknown-linux-gnueabihf-dev/include/blkid -I/nix/store/vf5042zi9xqv7gjrj17df5a249x9xkh8-glib-2.58.2-armv7l-unknown-linux-gnueabihf-dev/include/glib-2.0 -I/nix/store/kp65kyax63l43i1yavj2wzjf5rlh3w98-glib-2.58.2-armv7l-unknown-linux-gnueabihf/lib/glib-2.0/include -I/nix/store/vxmbjw65x5489z96bqggd5vwvqx7k4g5-gpgme-1.12.0-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/9b9583z5252pf2nihgqcv46anag1k1rf-libassuan-2.5.2-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/x6si6j0lmmp94k32wsv8k8ljv2kfcay8-libgpg-error-1.34-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/ikv8b4gsnhldy6mxzzmcsdk2cdwjc8r7-cryptsetup-2.1.0-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/1j5fmphfraiz5sp6ic36dxal6djvs37s-nss-3.42.1-armv7l-unknown-linux-gnueabihf-dev/include/nss -I/nix/store/nybzl0bnkmh9zzrgp2xkwglrrz9d7hab-nspr-4.20-armv7l-unknown-linux-gnueabihf-dev/include -g -O2 -Wall -W -Wcast-align -Wmissing-noreturn -Wnested-externs -Wpointer-arith -Wshadow -Wundef -Wwrite-strings -c python/volume_key_wrap.c  -fPIC -DPIC -o python/.libs/_volume_key_la-volume_key_wrap.o
  python/volume_key_wrap.c:149:11: fatal error: Python.h: No such file or directory
   # include <Python.h>
             ^~~~~~~~~~
  compilation terminated.
  make[2]: *** [Makefile:1091: python/_volume_key_la-volume_key_wrap.lo] Error 1
  make[2]: Leaving directory '/build/volume_key'
  make[1]: *** [Makefile:1234: all-recursive] Error 1
  make[1]: Leaving directory '/build/volume_key'
  make: *** [Makefile:745: all] Error 2
cannot build derivation '/nix/store/l66rj7kmv95nm0qh91va4fpkkl9q0r1d-libblockdev-2.20-armv7l-unknown-linux-gnueabihf.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/h4zn8sdb42zn052r2dhd5w0c2nj1520c-udisks-2.8.2-armv7l-unknown-linux-gnueabihf.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/0wcffjayikmczx3ixp370bfh06q21972-dbus-1.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/0a5nkxvzkiy9iap37iw477pls9sypjmz-hwdb.bin.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/6xz2h018p5h0p8yhd5g9vvvz1k5qjrkv-system-path.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/v17g2kk40xc8g651qv1046pbgc4qwzgz-system-units.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/hz446anikgdjyis3czs3xwxlyh5p1xbr-udev-rules.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/xqqfd7msas64r3a4gdnzh3p6l9wa2iyg-user-units.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/qyanxh77x51a71v5rcs6wrvd3gr6k7ka-nixos-system-nixos-on-arm-19.09pre-git.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/d51z0dpcdfg686i41lzldp6q3p9zi7zf-closure-info.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/gxlmn65finvkyr4fha25k1cnckxwkbqm-uEnv.txt.drv': 1 dependencies couldn't be built
cannot build derivation '/nix/store/hx8fsg0njsdh1v4wykk91p24aj3aqdg3-nixos-sd-image-19.09pre-git-armv7l-linux.img-armv7l-unknown-linux-gnueabihf.drv': 1 dependencies couldn't be built
[0 built (1 failed), 0.0 MiB DL]
error: build of '/nix/store/hx8fsg0njsdh1v4wykk91p24aj3aqdg3-nixos-sd-image-19.09pre-git-armv7l-linux.img-armv7l-unknown-linux-gnueabihf.drv' failed
```

deadloko, 2019-04-17:
Little more info on issue, it seems that directory containing "Python.h" does not gets included into build:
Native build:
```bash
swig -Wall -python -py3 -modern \
	-I. -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m \
	-o python/volume_key_wrap.c \
	./python/volume_key.i
./python/volume_key.i:387: Warning 321: 'open' conflicts with a built-in name in python
./python/volume_key.i:592: Warning 321: 'open' conflicts with a built-in name in python
/nix/store/h5f8yhs9cwxaq9kjimdwbxfhmijy5ysb-bash-interactive-4.4-p23/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.  -I/nix/store/hpkxfcggm5raf6h7fcwsghgz8944hqyq-util-linux-2.33.1-dev/include/blkid -I/nix/store/k6sfrjmgiddy4rmj66dgap8bd15r5g8p-glib-2.58.2-dev/include/glib-2.0 -I/nix/store/5fcij33f8q8yk127kvl45ydi9z7gmggj-glib-2.58.2/lib/glib-2.0/include -I/nix/store/r91g55nz9c0q6l99vxkp07hi3hjl43m9-gpgme-1.12.0-dev/include -I/nix/store/0v41s69z3x4myj78c8bd1sx1vz49sj31-libassuan-2.5.2-dev/include -I/nix/store/gwk3xrv20fnzfq6qi6rbvm1h9icbkh93-libgpg-error-1.34-dev/include -I/nix/store/qc2nv64j4s6spikbdy9y6n0qf821miw5-cryptsetup-2.1.0-dev/include -I/nix/store/08pz2m20jabxhwfhpv00xjc9jlmlgwvh-nss-3.42.1-dev/include/nss -I/nix/store/4hkvknkf04rzcc3r9w7akmk8q15pqjar-nspr-4.20-dev/include -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m  -Wno-unused-result -Wsign-compare  -DNDEBUG -g -fwrapv -O3 -Wall   -g -O2  -Wall -W -Wcast-align -Wmissing-noreturn -Wnested-externs -Wpointer-arith -Wshadow -Wundef -Wwrite-strings -MT python/_volume_key_la-volume_key_wrap.lo -MD -MP -MF python/.deps/_volume_key_la-volume_key_wrap.Tpo -c -o python/_volume_key_la-volume_key_wrap.lo `test -f 'python/volume_key_wrap.c' || echo './'`python/volume_key_wrap.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I/nix/store/hpkxfcggm5raf6h7fcwsghgz8944hqyq-util-linux-2.33.1-dev/include/blkid -I/nix/store/k6sfrjmgiddy4rmj66dgap8bd15r5g8p-glib-2.58.2-dev/include/glib-2.0 -I/nix/store/5fcij33f8q8yk127kvl45ydi9z7gmggj-glib-2.58.2/lib/glib-2.0/include -I/nix/store/r91g55nz9c0q6l99vxkp07hi3hjl43m9-gpgme-1.12.0-dev/include -I/nix/store/0v41s69z3x4myj78c8bd1sx1vz49sj31-libassuan-2.5.2-dev/include -I/nix/store/gwk3xrv20fnzfq6qi6rbvm1h9icbkh93-libgpg-error-1.34-dev/include -I/nix/store/qc2nv64j4s6spikbdy9y6n0qf821miw5-cryptsetup-2.1.0-dev/include -I/nix/store/08pz2m20jabxhwfhpv00xjc9jlmlgwvh-nss-3.42.1-dev/include/nss -I/nix/store/4hkvknkf04rzcc3r9w7akmk8q15pqjar-nspr-4.20-dev/include -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m -I/nix/store/gbwx5sc4w0i0ipnmsa0nsk88svs983vl-python3-3.7.2/include/python3.7m -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -g -O2 -Wall -W -Wcast-align -Wmissing-noreturn -Wnested-externs -Wpointer-arith -Wshadow -Wundef -Wwrite-strings -MT python/_volume_key_la-volume_key_wrap.lo -MD -MP -MF python/.deps/_volume_key_la-volume_key_wrap.Tpo -c python/volume_key_wrap.c  -fPIC -DPIC -o python/.libs/_volume_key_la-volume_key_wrap.o
```
Cross build:
```bash
swig -Wall -python -py3 -modern \
        -I.  \
        -o python/volume_key_wrap.c \
        ./python/volume_key.i
./python/volume_key.i:387: Warning 321: 'open' conflicts with a built-in name in python
./python/volume_key.i:592: Warning 321: 'open' conflicts with a built-in name in python
/nix/store/hllhp7xrhcrwd2b79rfcvwni0lbsfy0m-bash-4.4-p23/bin/bash ./libtool  --tag=CC   --mode=compile armv7l-unknown-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I.  -I/nix/store/dgijzk00lch1wg43c7ymj9qnbg2jv6gn-util-linux-2.33.1-armv7l-unknown-linux-gnueabihf-dev/include/blkid -I/nix/store/vf5042zi9xqv7gjrj17df5a249x9xkh8-glib-2.58.2-armv7l-unknown-linux-gnueabihf-dev/include/glib-2.0 -I/nix/store/kp65kyax63l43i1yavj2wzjf5rlh3w98-glib-2.58.2-armv7l-unknown-linux-gnueabihf/lib/glib-2.0/include -I/nix/store/vxmbjw65x5489z96bqggd5vwvqx7k4g5-gpgme-1.12.0-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/9b9583z5252pf2nihgqcv46anag1k1rf-libassuan-2.5.2-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/x6si6j0lmmp94k32wsv8k8ljv2kfcay8-libgpg-error-1.34-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/ikv8b4gsnhldy6mxzzmcsdk2cdwjc8r7-cryptsetup-2.1.0-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/1j5fmphfraiz5sp6ic36dxal6djvs37s-nss-3.42.1-armv7l-unknown-linux-gnueabihf-dev/include/nss -I/nix/store/nybzl0bnkmh9zzrgp2xkwglrrz9d7hab-nspr-4.20-armv7l-unknown-linux-gnueabihf-dev/include    -g -O2  -Wall -W -Wcast-align -Wmissing-noreturn -Wnested-externs -Wpointer-arith -Wshadow -Wundef -Wwrite-strings -c -o python/_volume_key_la-volume_key_wrap.lo `test -f 'python/volume_key_wrap.c' || echo './'`python/volume_key_wrap.c
libtool: compile:  armv7l-unknown-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I/nix/store/dgijzk00lch1wg43c7ymj9qnbg2jv6gn-util-linux-2.33.1-armv7l-unknown-linux-gnueabihf-dev/include/blkid -I/nix/store/vf5042zi9xqv7gjrj17df5a249x9xkh8-glib-2.58.2-armv7l-unknown-linux-gnueabihf-dev/include/glib-2.0 -I/nix/store/kp65kyax63l43i1yavj2wzjf5rlh3w98-glib-2.58.2-armv7l-unknown-linux-gnueabihf/lib/glib-2.0/include -I/nix/store/vxmbjw65x5489z96bqggd5vwvqx7k4g5-gpgme-1.12.0-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/9b9583z5252pf2nihgqcv46anag1k1rf-libassuan-2.5.2-armv7l-unknown-linux-gnueabihf-dev/include -I/nix/store/x6si6j0 [... cut ...] include -I/nix/store/1j5fmphfraiz5sp6ic36dxal6djvs37s-nss-3.42.1-armv7l-unknown-linux-gnueabihf-dev/include/nss -I/nix/store/nybzl0bnkmh9zzrgp2xkwglrrz9d7hab-nspr-4.20-armv7l-unknown-linux-gnueabihf-dev/include -g -O2 -Wall -W -Wcast-align -Wmissing-noreturn -Wnested-externs -Wpointer-arith -Wshadow -Wundef -Wwrite-strings -c python/volume_key_wrap.c  -fPIC -DPIC -o python/.libs/_volume_key_la-volume_key_wrap.o
python/volume_key_wrap.c:149:11: fatal error: Python.h: No such file or directory
```

illegalprime, 2019-04-18:
@deadloko sorry for the late response, the beaglebone image works on the latest master so this is really odd: https://travis-ci.org/illegalprime/nixos-on-arm/builds/518796537

What hash of this repo and nixpkgs do you have?

illegalprime, 2019-04-18:
This is the commit to fix volume_key ace1c6f85663dd48db3caac22668642c887a58b9 I guess I would double check your nixpkgs contains that hash

deadloko, 2019-04-18:
@illegalprime I am currently using last version of this repo and nixpkgs submodule.
I updated my nixos and tried enabling cachix arm cache. After that i managed to successfully build volume_key (actually it's just downloaded cached version). After that i disabled cachix, removed nix-store cache for package, and rebuilt it successfully. I will try and reproduce this error by fully rebuilding from scratch.

deadloko, 2019-04-19:
@illegalprime I am currently unable to reproduce this, even with full rebuild, so just closing this for now.

closed 2019-04-19 by deadloko

illegalprime, 2019-04-19:
@deadloko sorry about the heisenbug, but if you get your board working consider making a PR so other people can make use of it. best of luck!