illegalprime/nixos-on-arm #11: nixpkgs upstreaming

ElvishJerricco, 2019-05-02:
Is it reasonable to upstream all the nixpkgs-level work into nixpkgs proper? How much work would this be?

illegalprime, 2019-05-02:
@ElvishJerricco yea, absolutely, this is a goal of the project. Once a lot of boards / fixes are cleaned up I can send them upstream, some of the progress is monitored here:
https://github.com/illegalprime/nixos-on-arm#what-needs-to-be-done

That being said a lot of the fixes aren't cleaned up in a way nixpkgs might like, so that's the main work involved. If there's something specific you think is valuable let me know, otherwise feel free to cherry pick some commits from my branch and open a PR against upstream and @ me.

What's your use case for this project if you don't mind me asking? Also what boards would you be interested in seeing?

ElvishJerricco, 2019-05-03:
@illegalprime mostly I'm just intrigued :P But I may use this in a backup network. My desktop regularly backs up to an on-site external disk. I want to install a raspberry pi elsewhere for the sole purpose of also having backups in a geographically remote location, and I think it'd be awesome to deploy to it with NixOps and to cross compile the software with my far more powerful desktop.

closed 2019-09-16 by illegalprime

ElvishJerricco, 2019-09-19:
@illegalprime Why was this closed?