illegalprime/nixos-on-arm #15: Investigate Raspberry Pi 4

illegalprime, 2019-09-17:
is it in the upstream kernel?
how much work would t be to add it?
let's find out!

illegalprime, 2019-09-17:
didn't know upstream was already making good progress
https://github.com/NixOS/nixpkgs/pull/68265

It might be easy for u-boot to be patched, will investigate more.

samueldr, 2019-09-20:
U-Boot will get support after the mainline kernel has initial support, from what I remember reading. Something about waiting on the kernel for the device tree source files, as they go `kernel` → `u-boot` or something like that.

illegalprime, 2019-09-26:
Just saw this, thanks for the info/link!

jpraczyk, 2019-10-22:
Just an fyi @illegalprime, it seems that u-boot 2019.10 already supports rpi 4.

samueldr, 2019-10-22:
Yes it should, I updated u-boot, and also opened a PR with the added u-boot for rpi4, but..

 * This is not tested
 * It lacks the changes to the generic image

https://github.com/NixOS/nixpkgs/pull/70796

Changes should be relatively trivial, I should be able to get back to it by November, unless someone else got it going before I have the time to work on that.