illegalprime/nixos-on-arm #22: Missing options error

DanLipsitt, 2019-10-06:
I'm new to Nix so forgive me if I'm missing something obvious, but I get the following errors on both MacOS and NixOS running in VirtualBox:

```console
$ nix build -f . -I nixpkgs=nixpkgs  -I machine=machines/beaglebone -I image=images/mini
error: The option `documentation' defined in `/vagrant/images/mini' does not exist.
(use '--show-trace' to show detailed location information)

$ nix build -f . -I nixpkgs=nixpkgs -I machine=machines/beaglebone -I image=images/base
error: The option `boot.loader.generic-extlinux-compatible.dtbs' defined in `/vagrant/machines/beaglebone' does not exist.
(use '--show-trace' to show detailed location information)
```

SlothOfAnarchy, 2019-11-11:
I got the same error, I think the option got moved or renamed and now has to be set in `nixpkgs.crossSystem.dtb` now (a string instead of a list), like it has been done in https://github.com/illegalprime/nixos-on-arm/blob/master/machines/toradex_apalis_imx6/system.nix#L21

I'm not 100% sure though, because I still have other errors.

SlothOfAnarchy, 2019-11-17:
@DanLipsitt  see https://github.com/illegalprime/nixos-on-arm/pull/24